RAKWireless/rak_common_for_gateway #8: Boot failure after rak_common_for_gateway install: hangs with mmc1: new high speed ...

hdtodd, 2020-01-14:
I'm trying to set up a LoRa gateway on a previously-installed RPi-3B+ Buster system.  I just received a new RAK2245.

I2C and SPI enabled.

I installed the gateway software on a RPi-3B+ running Buster; updated/upgrades OS software before gateway install.  I followed the gateway install instructions (./install.sh), and the installation seemed to have completed successfully; I identified the hardware as RAK2245.  Reset the hostname to my device's hostname using raspi-config and rebooted.  (first reboot after installing 2245 hardware)

Boot now hangs.  The startup log screen flows normally for a couple of seconds, then the system hangs with:
            "mmc1:  new high speed SDIO card at address 0001"
This was preceded a few lines earlier with a message
            "mmc1: queuing unknown CIS tuple 0x80 (2 bytes)"
followed by the same message with 3, 3, and 7 bytes.

Powered off and removed the 2245 hardware, thinking that might be the problem.  NOPE!  With the hardware removed, the system still hangs at the same place with the same messages!

So, this appears to be a software issue, not a hardware issue.  Anyone else see this?  Any idea of a parameter I can change in gateway config files to let me boot up again, with or without the 2245 installed?

hdtodd, 2020-01-14:
Restored prior /boot/config.txt and system now boots again.

Comparing the working config.txt with the gateway-installed modified version, helps narrow down the configuration change that's causing the problem.  "<" is working; ">" is failing version.

I have a feeling it's the "> dtparam=i2c1 ..." line that's causing the problem (I've had SPI and I2C on before).  

Any suggestions as to what in that line should be changed to make the config bootable?  Or is it some other parameter?

Does Bluetooth have to be disabled for the gateway code?

< dtparam=i2c_arm=off
---
> dtparam=i2c_arm=on
48c48,49
< dtparam=spi=off
---
> dtparam=spi=on
> dtparam=i2c1=on, dtparam=i2c_arm_baudrate=100000
56,70c57,59
< dtparam=audio=off
< 
< [pi4]
< # Enable DRM VC4 V3D driver on top of the dispmanx display stack
< dtoverlay=vc4-fkms-v3d
< max_framebuffers=2
< 
< [all]
< #dtoverlay=vc4-fkms-v3d
< start_x=0
< enable_uart=0
< 
< #disable polling for sdcard for USB booting
< dtoverlay=sdtweak,poll_once
< 
---
> dtparam=audio=on
> dtoverlay=pi3-disable-bt
> dtoverlay=dwc2

hdtodd, 2020-01-14:
Found it ... the "dtoverlay=dwc2" line in config.txt was causing the system to hang.  May have been because I already had "modules-load=dwc2" in cmdline.txt for other reasons.

RAKWireless, 2020-01-15:
@hdtodd "modules-load = dwc2" is added to support Raspberry Pi 0 (not pi0 w) to connect to the network via USB. In the next version we will remove "modules-load = dwc2" and remove support for Raspberry Pi 0.

closed 2020-02-09 by hdtodd