RAKWireless/rak_common_for_gateway #9: Install fails due to missing packages

rogerdahl, 2020-01-28:
The current  `chirpstack/install.sh` references non-existing versions of the `.deb` packages.

The following locations are currently valid (version numbers must be changed in two places in the script, sigh).
```
wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-application-server/chirpstack-application-server_3.7.0_linux_armv7.deb
wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-gateway-bridge/chirpstack-gateway-bridge_3.6.0_linux_armv7.deb
wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-network-server/chirpstack-network-server_3.6.0_linux_armv7.deb
```
Instead of hard coded versions, the installer should reference `latest`, which should be a redirect to latest version on the site.

titansmc, 2020-02-16:
They have already changed, The way to do it should be to check which are the available packages from:
```
https://artifacts.chirpstack.io/packages/3.x/deb/dists/stable/main/binary-arm64/Packages
```

saket424, 2020-02-21:
edit the cirpstack/install.sh as follows

<pre>
pi@rak-gateway:~/rak_common_for_gateway/chirpstack $ diff install.sh.orig install.sh
67,72c67,72
< #apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1CE2AFD36DBCCA00
< #sudo echo "deb https://artifacts.chirpstack.io/packages/3.x/deb stable main" | sudo tee /etc/apt/sources.list.d/chirpstack.list
< #apt update
< #apt install chirpstack-network-server
< #apt install chirpstack-gateway-bridge
< #apt install chirpstack-application-server
---
> apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1CE2AFD36DBCCA00
> sudo echo "deb https://artifacts.chirpstack.io/packages/3.x/deb stable main" | sudo tee /etc/apt/sources.list.d/chirpstack.list
> apt update
> apt install chirpstack-network-server
> apt install chirpstack-gateway-bridge
> apt install chirpstack-application-server
74,78c74,78
< rm *.deb -f
< #3.2 download chirpstack packages
< wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-application-server/chirpstack-application-server_3.5.1_linux_armv7.deb
< wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-gateway-bridge/chirpstack-gateway-bridge_3.4.1_linux_armv7.deb
< wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-network-server/chirpstack-network-server_3.6.0_linux_armv7.deb
---
> #rm *.deb -f
> ##3.2 download chirpstack packages
> #wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-application-server/chirpstack-application-server_3.5.1_linux_armv7.deb
> #wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-gateway-bridge/chirpstack-gateway-bridge_3.4.1_linux_armv7.deb
> #wget https://artifacts.chirpstack.io/packages/3.x/deb/pool/main/c/chirpstack-network-server/chirpstack-network-server_3.6.0_linux_armv7.deb
81,83c81,83
< dpkg -i chirpstack-application-server_3.5.1_linux_armv7.deb
< dpkg -i chirpstack-gateway-bridge_3.4.1_linux_armv7.deb
< dpkg -i chirpstack-network-server_3.6.0_linux_armv7.deb
---
> #dpkg -i chirpstack-application-server_3.5.1_linux_armv7.deb
> #dpkg -i chirpstack-gateway-bridge_3.4.1_linux_armv7.deb
> #dpkg -i chirpstack-network-server_3.6.0_linux_armv7.deb

</pre>

saket424, 2020-02-21:
If all goes well with cirpstack install, you should see this

pi@rak-gateway:~/rak_common_for_gateway/chirpstack $ dpkg -l | grep chirp
ii  chirpstack-application-server  3.8.0                               armhf        ChirpStack Application Server
ii  chirpstack-gateway-bridge      3.7.1                               armhf        ChirpStack Gateway Bridge
ii  chirpstack-network-server      3.7.0                               armhf        ChirpStack Network Server

ZhuQI77, 2020-03-12:
chirpstack always deletes historical versions.
In the next version we will  add the chirpstack's repository to the repository list to fix this bug.

RAKWireless, 2020-03-12:
chirpstack always deletes historical versions.
In the next version we will add the chirpstack's repository to the repository list to fix this bug.

closed 2020-03-24 by RAKWireless