RAKWireless/rak_common_for_gateway #12: Discrepancy between LUT sources and obtained values.

sophiekovalevsky, 2020-03-16:
Hi RAKWireless team:

I'm providing support for PiHat 2245 to work @ AS923 ISM band and I stumbled across some issues as regards to the LUT that you provide. Please see https://github.com/brocaar/chirpstack-gateway-os/issues/41

Based on the issue provided we are moving towards using [this](https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2245/global_conf/global_conf.as_923.json) config option since it would be easier to get notifications about recent updates. However, I do have a couple of questions about it. 

1. Why there are several configuration files? 

2. What were the electrical and other related variable conditions while taking measurements in https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2245/global_conf/global_conf.as_923.json)? Since measured power level differs from the ones are stay there.

Please take a look at the following image when PA = 1, Mix = 8 and Dig = 0

[image]

My results offset 2.3 dBm which is huge compared to what it is says: 3 dBm.

RAKWireless, 2020-03-17:
@sophiekovalevsky 
Thank you very much for choosing RAK.

> Why there are several configuration files?
Do you mean why there are several global_conf.xxxx_json files in the https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2245/global_conf/ directory?

We have noticed the problem of lut you feedback before. Our team is currently optimizing the parameters of lut, and we will update as soon as the test is complete.

sophiekovalevsky, 2020-03-17:
@RAKWireless 

I mean, why there are several configuration files in the following source where values don't match. 

1. [RAK2245-RAK831-LoRaGateway-RPi-Raspbian-OS](https://github.com/RAKWireless/RAK2245-RAK831-LoRaGateway-RPi-Raspbian-OS/blob/master/lora/global_conf.json)

2. [Global configuration for AS 923 at RAK Common for Gateway](https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2245/global_conf/global_conf.as_923.json)

3. [Transmitter RF Characteristics](https://doc.rakwireless.com/datasheet/rakproducts/rf-characteristics-rak2245-pi-hat#transmitter-rf-characteristics])

Then, based on your response, I can assume that the most updated source from where I can get the parameters would be 2, right? 

Would be great if you can enclose the additional parameters about how you got the LUTs values, so that measurement conditions are clear. Otherwise, this kind of issues may be hidden preventing to fix  it soon.

RAKWireless, 2020-03-19:
@sophiekovalevsky 
We will update the global_conf.json file next week.

We will notify you as soon as the update is complete.

Thank you again for your valuable suggestions.

sophiekovalevsky, 2020-03-19:
@RAKWireless thanks in advance for letting know.

RAKWireless, 2020-03-20:
@sophiekovalevsky 
Hello！
I have updated lut configuration for dev branch 800 band and 900 band.
You can get the latest global_conf.json from the dev branch.
We hope RAK can bring you a better product experience.

sophiekovalevsky, 2020-03-20:
@RAKWireless thanks so much for having a look at these. 

I have made some comments in the lut configurations for AS923, let me know your thoughts.

I took again some measurements with your new changes and I still not getting the same values than yours. Please find more details below:

Measurements were made using RAK2245 Ver A. at T ~ 25 C, Vdd ~ 5 V @ 2 A, BW: 125 kHz, SF: 7 @ 923.2 MHz using a continuous wave transmission.

| LUT index | PA Gain | Mix Gain | Dig Gain | Tx Power by RAK (dBm) | Measured (dBm)
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------- |
| 0 | 0 | 8  | 0 | -6 |  -6  |
| 1 | 0 | 11 | 0  | -3 |  -2.5  |
| 2 | 0 | 14 | 0  | 0 |  0  |
| 3 | 1 | 8 | 0  | 3 |  5  |
| 4 | 1 | 10 | 0  | 6 |  8.4  |
| 5 | 1 | 14 | 0  | 10 |  13.6  |
| 6 | 2 | 8 | 0  | 11 |  10.5  |
| 7 | 2 | 8 | 0  | 12 |  10.5 |
| 8 | 2 | 9 | 0  | 13 |  12.5 |
| 9 | 2 | 10 | 0  | 14 |  14 |
| 10 | 2 | 11 | 0  | 16 |  16 |
| 11 | 3 | 8 | 0  | 20 |  20 |
| 12 | 3 | 9 | 0  |  23 |  21.7 |
| 13 | 3 | 11 | 0  | 25 |  24.8 |
| 14 | 3 | 13 | 0  | 26 |  26.6 |
| 15 | 3 | 14 | 0  | 27 |  27 |

If there is something that I am missing here that would explain the why I am not getting same results? I would appreciate your feedback on this. Thanks so much.

RAKWireless, 2020-03-23:
@sophiekovalevsky
Hello！
Tx Power by RAK (dBm) is the target power. The real power(Measured (dBm)) is impossible the same as it.In fact,each board's tx power will be different even with the same parameters because of components' discrete characteristic.

sophiekovalevsky, 2020-03-23:
@RAKWireless understood. Thanks.

closed 2020-03-23 by sophiekovalevsky