RAKWireless/rak_common_for_gateway #13: RAK2246 tx gain table configuration

brocaar, 2020-03-22:
I'm adding support for the RAK2246 module to the https://github.com/brocaar/chirpstack-concentratord project (https://github.com/brocaar/chirpstack-concentratord/pull/5).

I noticed that the [configuration](https://github.com/RAKWireless/rak_common_for_gateway/tree/master/lora/rak2246/global_conf) for the EU868 and US915 band have 8 entries in the tx gain table, the other bands have 16 entries in the tx gain table. The 8 entries are the same for the EU868 and US915 band and also the 16 entries for the other bands are the same.

Is this correct?

This might be related to https://github.com/RAKWireless/rak_common_for_gateway/issues/12 which is about the RAK2245.

RAKWireless, 2020-03-23:
Hi Brocaar,
RAK2246 Pi HAT and RAK2245 Pi HAT have different FEM scenario.

For RAK2245 Pi HAT:
1.	The PA used on RAK2245 Pi HAT is RFPA0133 which works over a range of frequencies
from 380MHz to 960MHz.It also has two PINs(G8 and G16) to control PA’s gain.
 
[image]

2.	MIX : Radio Tx mixer gain trim, [8:15]
3.	DIG: Digital gain trim, [0:3], 0:1, 1:7/8, 2:3/4, 3:1/2
There are four parameters(G8,G16,MIX,DIG) to set entry. Usually, DIG defaults to 0. So RAK2245 Pi HAT has 16 entries in the tx gain table.

For RAK2246 Pi HAT:
1.	The PA used on RAK2246 Pi HAT is SKY66422-11 which is an integrated FEM and works over a range of frequencies from 860MHz to 930MHz.It’s gain is fixed and flat.
2.	MIX : SX125X Radio Tx mixer gain trim, [8:15]
3.	DIG: Digital gain trim, [0:3], 0:1, 1:7/8, 2:3/4, 3:1/2
There are two parameters(MIX,DIG) to set entry. Usually, DIG defaults to 0. So RAK2246 Pi HAT has 8 entries in the tx gain table. You can get 16 entries in the tx gain table if you change DIG’s value also.

Do you have to have 16 entries in the tx gain table?

brocaar, 2020-03-23:
Thanks for your response!

> Do you have to have 16 entries in the tx gain table?

This is not a requirement for me, but I did notice that most of the bands do have 16 entries, except the EU868 and US915 bands. Also these 16 entries are exactly the same as https://github.com/lora-net/packet_forwarder/blob/master/lora_pkt_fwd/cfg/global_conf.json.PCB_E286.EU868.basic#L86.

This made me wonder which configuration is correct and potentially which configuration comes from the Semtech example configuration.

Could you explain why the EU868 and US915 configuration is the same, but not for example the EU868 and RU864?

RAKWireless, 2020-03-23:
@brocaar 
We use the configuration file provided by ttn (https://github.com/TheThingsNetwork/gateway-conf) by default.

The power calibration of EU868 and US915 was completed a few days ago, and the power calibration tests of other frequency bands will be completed in recent days. After completion, it will stay with EU868 and US915, and lut keeps eight.

In the future, more optimizations will be made to the power of rak2246 to keep lut at 16.

brocaar, 2020-03-23:
Ok, looking forward to the calibration values of the other bands :) The reason why I'm asking for this is because incorrect configuration could mean that a gateway does not perform as expected. The user might think it is transmitting at x dBm, while it is not (it might be under performing).

With the Concentratord model configuration, I hope to collect all the correct calibration values for all vendors and gateway models, so that when a user sets it to e.g. `rak_2246_eu868`, it can be assured that the correct calibration is applied.

RAKWireless, 2020-03-23:
@brocaar 
Thank you very much for your attention and suggestions, this will make our products better and better.

RAKWireless, 2020-03-24:
@brocaar 
We have updated the global_conf.json file for rak2246 on the dev branch. rak2246 does not have low frequency products at the moment, so please ignore cn470 and eu433 at this stage.

brocaar, 2020-03-24:
Thanks for the updates.

closed 2020-03-24 by brocaar