RAKWireless/rak_common_for_gateway #15: RAK833 - failed to start the concentrator

Dodoooh, 2020-03-27:
Unfortunately, it's not working for me

I have a USB adapter for the RAK833 which is directly connected to the Raspberry. I always get the following error message: 
**failed to start the concentrator**

I have already tested it with a Raspberry Pi 3 and 4.
As well as with the distribution buster and stretch 

```
Mar 27 22:50:49 rak-gateway systemd[1]: ttn-gateway.service: Service RestartSec=5s expired, scheduling restart.
Mar 27 22:50:49 rak-gateway systemd[1]: ttn-gateway.service: Scheduled restart job, restart counter is at 6.
Mar 27 22:50:49 rak-gateway systemd[1]: Stopped The Things Network Gateway.
Mar 27 22:50:49 rak-gateway systemd[1]: Started The Things Network Gateway.
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: *** Beacon Packet Forwarder for Lora Gateway ***
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: Version: 4.0.1
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: *** Lora concentrator HAL library version info ***
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: Version: 5.0.1;
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: ***
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Little endian host
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: found global configuration file global_conf.json, parsing it
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: lorawan_public 1, clksrc 1
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: no configuration for LBT
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: antenna_gain 0 dBi
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Configuring TX LUT with 16 indexes
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: gateway MAC address is configured to 0000000000000000
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: server hostname or IP address is configured to "router.eu.thethings.network"
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: upstream port is configured to "1700"
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: downstream port is configured to "1700"
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: downstream keep-alive interval is configured to 10 seconds
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: statistics display interval is configured to 30 seconds
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: upstream PUSH_DATA time-out is configured to 100 ms
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with a valid CRC will be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with a CRC error will NOT be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with no CRC will NOT be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: Auto-quit after 20 non-acknowledged PULL_DATA
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: found local configuration file local_conf.json, parsing it
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: redefined parameters will overwrite global parameters
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: local_conf.json does not contain a JSON object named SX1301_conf
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: gateway MAC address is configured to XXXXXXXXXXX
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with a valid CRC will be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with a CRC error will NOT be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: INFO: packets received with no CRC will NOT be forwarded
Mar 27 22:50:50 rak-gateway ttn-gateway[792]: ERROR: [main] failed to start the concentrator
Mar 27 22:50:50 rak-gateway systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 22:50:50 rak-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
```

RAKWireless, 2020-03-30:
`Please select your gateway model:
  *	1.RAK831
  *	2.RAK2245
  *	3.RAK7243/RAK7244 no LTE
  *	4.RAK7243/RAK7244 with LTE
  *	5.RAK833(USB)
  *	6.RAK2247(USB)
  *	7.RAK833(SPI)
  *	8.RAK2247(SPI)
  *	9.RAK2246
  Please enter 1-9 to select the model:`
Did you choose "5.RAK833 (USB)" during installation?
Can you send me the /opt /ttn-gateway/ compression?
qi.zhu@rakwireless.com

RAKWireless, 2020-03-30:
make sure the mini-pcie PERST# signal(pin 22) pulled down (default high will cause rak833 function error).

Dodoooh, 2020-03-31:
> `Please select your gateway model:
> 
>     * 1.RAK831
> 
>     * 2.RAK2245
> 
>     * 3.RAK7243/RAK7244 no LTE
> 
>     * 4.RAK7243/RAK7244 with LTE
> 
>     * 5.RAK833(USB)
> 
>     * 6.RAK2247(USB)
> 
>     * 7.RAK833(SPI)
> 
>     * 8.RAK2247(SPI)
> 
>     * 9.RAK2246
>       Please enter 1-9 to select the model:`
> 
> 
> Did you choose "5.RAK833 (USB)" during installation?
> Can you send me the /opt /ttn-gateway/ compression?
> Can you send me the information displayed after the gateway-version command is executed?
> [qi.zhu@rakwireless.com](mailto:qi.zhu@rakwireless.com)

hey 
i sent you an email with all the details. 
Thanks for your help!

closed 2020-04-21 by RAKWireless