RAKWireless/rak_common_for_gateway #16: Chirpstack install fails, .deb package links return 404

Jickelsen, 2020-04-02:
The versions linked in the install script here

https://github.com/RAKWireless/rak_common_for_gateway/blob/7f110c2f00831a33be72f378b7129cae4b2eb906/chirpstack/install.sh#L54

are no longer available. The version numbers need to be updated. I do wish Chirpstack would keep older versions up though...

Jickelsen, 2020-04-03:
The versions specified in the install script can now be found here:
https://artifacts.chirpstack.io/downloads/chirpstack-application-server/chirpstack-application-server_3.5.1_linux_armv7.deb
https://artifacts.chirpstack.io/downloads/chirpstack-gateway-bridge/chirpstack-gateway-bridge_3.4.1_linux_armv7.deb
https://artifacts.chirpstack.io/downloads/chirpstack-network-server/chirpstack-network-server_3.6.0_linux_armv7.deb

Jickelsen, 2020-04-06:
I realise now this is one a problem for the latest release, 4.1.1, and has been fixed in 4.2. Please consider creating a release for 4.2.

RAKWireless, 2020-04-08:
@Jickelsen 
Thank you for your suggestion, we have created a release for 4.2.0.

closed 2020-04-08 by RAKWireless