RAKWireless/rak_common_for_gateway #17: getting a rakk2245 runbning with othger packet forwarders/gateways

dolfandringa, 2020-05-06:
I am using a rakk2245 on an orangepi zero (running armbian) instead of a raspberry pi (raspbian). I also prefer not to use prebuilt images foir my gateway but build my own. I managed to get it to work quite easily with your version of libloragw and the packet forwarder you supply. I only needed toi change the SPIDEV path to `/dev/sidev1.0` (instead of 0.0) and change the GPIO SX1301 RESET pin to 1 instead of 17. 

But I would actually prefer to use a different gateway as well. I am messing around with basic station right now. It also uses libloragw 5.0.1, just like you do. So my question is what I would need to do to get it to work. Right now I am getting issues with the SPI communication. I see the SPI speed is different (2MHz instead of 8 on libloragw). I changed that already in my basicstation libloragw version and I changed the GPIO RESET pin and made sure to reset the radio, but still I am getting an error `lgw_start:863: ERROR: CALIBRATION FAILURE (STATUS = 119)`. Any ideas how to fix these? I would be happy to contribute RAK2245 support to basic station if I mange to succeed.

dolfandringa, 2020-05-06:
This is the log that basicstation is giving (with libloragw DEBUG flags on):

```
2020-05-06 01:44:28.028 [SYS:INFO] Logging     : stderr (maxsize=10000000, rotate=3)
2020-05-06 01:44:28.028 [SYS:INFO] Station Ver : 2.0.3(rpi/std) 2020-05-06 01:29:59
2020-05-06 01:44:28.028 [SYS:INFO] Package Ver : (null)
2020-05-06 01:44:28.028 [SYS:INFO] proto EUI   : 242:58ff:fef7:bf4b     (station.conf file routerid.txt)
2020-05-06 01:44:28.028 [SYS:INFO] prefix EUI  : ::1    (builtin)
2020-05-06 01:44:28.028 [SYS:INFO] Station EUI : 242:58ff:fef7:bf4b
2020-05-06 01:44:28.028 [SYS:INFO] Station home: ./     (builtin)
2020-05-06 01:44:28.029 [SYS:INFO] Station temp: /var/tmp/      (builtin)
2020-05-06 01:44:28.029 [SYS:WARN] Station in NO-CUPS mode
2020-05-06 01:44:28.231 [TCE:INFO] Starting TC engine
2020-05-06 01:44:28.232 [TCE:INFO] Connecting to INFOS: ws://127.0.0.1:3001
2020-05-06 01:44:28.234 [TCE:INFO] Infos: 0242:58ff:fef7:bf4b 0242:58ff:fef7:bf4b ws://127.0.0.1:3001/gateway/024258fffef7bf4b
2020-05-06 01:44:28.235 [AIO:ERRO] Recv failed: NET - Connection was reset by peer
2020-05-06 01:44:28.236 [TCE:VERB] Connecting to MUXS...
2020-05-06 01:44:28.238 [TCE:VERB] Connected to MUXS.
Accessing concentrator reset pin through GPIO1...
lgw_board_setconf:449: Note: board configuration; lorawan_public:1, clksrc:1
lgw_rxrf_setconf:498: WARNING: NOT A VALID TX NOTCH FILTER FREQUENCY [126000..250000]Hz
lgw_rxrf_setconf:510: Note: rf_chain 0 configuration; en:1 freq:904300000 rssi_offset:-166.000000 radio_type:2 tx_enable:1 tx_notch_freq:0
lgw_rxrf_setconf:498: WARNING: NOT A VALID TX NOTCH FILTER FREQUENCY [126000..250000]Hz
lgw_rxrf_setconf:510: Note: rf_chain 1 configuration; en:1 freq:905300000 rssi_offset:-166.000000 radio_type:2 tx_enable:1 tx_notch_freq:0
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 0 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 1 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 2 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 3 configuration; en:1 freq:200000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:400000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:-400000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:-200000 SF_mask:0x7e
lgw_rxif_setconf:629: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:0 SF_mask:0x7e
lgw_rxif_setconf:603: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:300000 bw:1 dr:4
lgw_rxif_setconf:537: Note: if_chain 9 disabled
Note: SPI port opened and configured ok
Note: SPI read success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI write success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
Note: SPI read success
Note: SPI write success
[... 258 lines cut ...]
lgw_start:863: ERROR: CALIBRATION FAILURE (STATUS = 127)
2020-05-06 01:44:28.243 [SYS:VERB] ./deps/lgw/platform-rpi/reset_lgw.sh start 1: Forked, waiting...
2020-05-06 01:44:28.882 [SYS:INFO] Process ./deps/lgw/platform-rpi/reset_lgw.sh start 1 (pid=6782) completed
2020-05-06 01:44:28.900 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=904300000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=905300000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  0: enable=1 rf_chain=0 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  1: enable=1 rf_chain=0 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  2: enable=1 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  3: enable=1 rf_chain=0 freq=200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  4: enable=1 rf_chain=0 freq=400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  5: enable=1 rf_chain=1 freq=-400000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  6: enable=1 rf_chain=1 freq=-200000 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  7: enable=1 rf_chain=1 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  8: enable=1 rf_chain=0 freq=300000 bandwidth=1 datarate=4 sync_word=0/0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 ifchain  9: enable=0 rf_chain=0 freq=0 bandwidth=0 datarate=0 sync_word=0/0
2020-05-06 01:44:28.901 [RAL:VERB] SX1301 LBT not enabled
2020-05-06 01:44:28.901 [RAL:INFO] Station device: /dev/spidev1.0 (PPS capture disabled)
2020-05-06 01:44:31.795 [RAL:ERRO] lgw_start failed: /dev/spidev1.0
2020-05-06 01:44:31.795 [any:ERRO] Closing connection to muxs - error in s2e_onMsg
2020-05-06 01:44:31.806 [TCE:VERB] Connection to MUXS closed in state -1
2020-05-06 01:44:31.806 [TCE:INFO] INFOS reconnect backoff 10s (retry 1)
Note: SPI write success
Note: SPI port closed
```

dolfandringa, 2020-05-06:
See also my issue with basicstation: https://github.com/lorabasics/basicstation/issues/60

RAKWireless, 2020-05-08:
@dolfandringa 
Hello!
We have tested rak831, rak2247 spi version and rak2247 usb version on orangepi3, hope to help you.

dolfandringa, 2020-05-08:
> @dolfandringa
> Hello!
> We have tested rak831, rak2247 spi version and rak2247 usb version on orangepi3, hope to help you.

I am asking about the rak2245. Could you please look at the question and answer that?

dolfandringa, 2020-05-14:
Can you please take a look at the ticket https://github.com/lorabasics/basicstation/issues/60 where the people are willing to help from the side of basicstation. Maybe if you can assist as well, we can resolve this, and you can report the rakk2245 to also work with basicstation.

tonysmith55, 2020-05-14:
@dolfandringa , I notice in your log from 8 days ago, you have both radios activated to Transmit.  Normally only one radio is active to transmit, both are active to receive.  I have no idea if this is the cause of your problem.
SX1301 rxrfchain 0: enable=1 freq=904300000 rssi_offset=-166.000000 type=2 **tx_enable=1** tx_notch_freq=0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=905300000 rssi_offset=-166.000000 type=2 **tx_enable=1** tx_notch_freq=0

Normally only rxrfchain 0 is active, rxrfchain 1 is normally not active.

you could try
SX1301 rxrfchain 0: enable=1 freq=904300000 rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
2020-05-06 01:44:28.900 [RAL:VERB] SX1301 rxrfchain 1: enable=1 freq=905300000 rssi_offset=-166.000000 type=2 **tx_enable=0** tx_notch_freq=0

Tony

dolfandringa, 2020-05-18:
thanks @tonysmith55 for the answer, but how can I "try" that. I posted logs of what basicstation is logging, which uses libloragw just like the rak2245 packet forwarder does. And the basicstation version is logging a "Calibration error" while the rak2245 packet forwarder works fine. So if that error is caused by both chains being active, how do I change that?

dolfandringa, 2020-05-18:
OK, lol. Youre suggestion @tonysmith55 made me look at my config file. It turns out I did enable TX on both radio 0 and radio 1 in the basicstation `station.conf`. Switching it off on radio 1 did remove the calibration error. It hasn't fixed all my issues, but I think right now it is between chirpstack and basictation, and has nothing to do with the rakk22455. Thanks for the help. And I'll close this ticket for now.

closed 2020-05-18 by dolfandringa