RAKWireless/rak_common_for_gateway #22: global_conf.json: No such file or directory

nedhorning, 2020-06-29:
I'm getting this error message: can't read /opt/ttn-gateway/packet_forwarder/lora_pkt_fwd/global_conf.json: No such file or directory

when I run sudo gateway-config” => Setup RAK Gateway LoRa concentrator for a Chirpstack server. I was running this last week without any problem so I expect the update yesterday introduced an error.

Scobber, 2020-06-29:
are you using chirpstack or things network?
the update yesterday seemed to address only a things network bug.

Scobber, 2020-06-29:
and I would assume that updating the software probably deleted your configuration. and reset it all to default.

nedhorning, 2020-06-29:
> EXTERNAL SENDER

I am using ChirpStack. I get the error after setting the “SERVIR_IP” 
to“127.0.0.1” in the "ChirsStack Channel-plan configuration"

For what it's worth, I set this up using a fresh RPi image

Ned

Scobber, 2020-06-30:
can you post that file you modified. 
also if you can also post the output of /var/log/daemon.log

perhaps its not validating correctly. with lorawan you have two files in that folder, the global and a local conf. the local overwrites the json dictionary from global

nedhorning, 2020-06-30:
I attached the daemon.log file.

I didn't modify any files on my side. This was a clean install using a 
freshly burned RPi OS image. In my original post when I wrote "I expect 
the update yesterday introduced an error." I was referring to the  
update on the rak_common_for_gateway repo. Sorry for not being clear. I 
have no idea if that was the issue but it was the only change I am aware of.

Ned

nedhorning, 2020-07-01:
Here's something else that might be related. When I run the install.sh script I get the following messages in the last two lines in the terminal after the "Install ChirpStack success!" message. I don't recall if I got these messages in the past.

./install_normal.sh: line 56: RAK7243: command not found
./install_normal.sh: line 61: popd: directory stack empty

Scobber, 2020-07-01:
paste the output of 

sudo gateway-config

nedhorning, 2020-07-02:
> EXTERNAL SENDER
>
> paste the output of
>
> sudo gateway-config

For what it's worth I do not have this issue when using the 
"RAK7243_Latest_Firmware" image or the image for ChirpStack Gateway OS.

Thanks for looking into this.

RAKWireless, 2020-07-08:
> Here's something else that might be related. When I run the install.sh script I get the following messages in the last two lines in the terminal after the "Install ChirpStack success!" message. I [...]

Hello @nedhorning . 
I'm sorry I discovered this bug so late. I have fixed it today.

closed 2020-07-22 by RAKWireless