RAKWireless/rak_common_for_gateway #33: Problem using RAK2247

SWoto, 2020-10-06:
Hello everyone,  

I'm currently trying to use the RAK2247 with he CM3+, running the Raspberry OS Lite but a few seconds/minutes(?) after the installation is completed i lose my SSH connection with it and cannot get in anymore. Also, the `gateway_ID` in the `global_conf.json` is still "0000000000000000" instead of that mixture of my ethernet mac.

**My Configuration:**
- (CM3+)[https://www.raspberrypi.org/products/compute-module-3-plus/]
- (Raspberry OS Lite)[https://www.raspberrypi.org/downloads/raspberry-pi-os/]
- Connected through SSH
- Network access using cable (eth0), the board has no WiFi.

**Problem:**
After I lost the connection, looks like the board is rebooting itself after it shutdown (by itself). This is the ping result:
```
$ ping 192.168.0.114
PING 192.168.0.114 (192.168.0.114): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
ping: sendto: No route to host
Request timeout for icmp_seq 4
ping: sendto: Host is down
[...]
Request timeout for icmp_seq 21
ping: sendto: Host is down
Request timeout for icmp_seq 22
ping: sendto: Host is down
Request timeout for icmp_seq 23
Request timeout for icmp_seq 24
Request timeout for icmp_seq 25
Request timeout for icmp_seq 26
Request timeout for icmp_seq 27
Request timeout for icmp_seq 28
ping: sendto: No route to host
Request timeout for icmp_seq 29
ping: sendto: Host is down
[...]
```

**Logs:**
```
$ uname -a
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
```

```
$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.114  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::c948:efaf:719:404f  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:80:f4:c5  txqueuelen 1000  (Ethernet)
        RX packets 4079  bytes 1784545 (1.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4261  bytes 3866857 (3.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2728  bytes 5129684 (4.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2728  bytes 5129684 (4.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

```
$ lsusb
Bus 001 Device 005: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
[...]
[... 431 lines cut ...]
Device "wlan0" does not exist.
Install ap success!
[...]
WARNING: ChirpStack is not support for this rpi model.
[...]
```

Here is the image showing no gateway_ID.
[image]

**Thinking...**
In my opinion, this might be related to this:
> step5 : Wait a moment and the installation is complete.If your gateway uses a wired connection, please reconfigure the LAN's IP address using "sudo gateway-config" after the installation is complete.
But, what should I do?
[image]

**Steps to reproduce:**
1. Fresh install Raspberry OS Lite
2. Connect the raspberry to the router using an Ethernet Cable
3. Connect to it through SSH
4. `sudo apt-get update && sudo apt-get upgrade -y`, then reboot
5. Follow (Installation Procedure)[https://github.com/RAKWireless/rak_common_for_gateway#installation-procedure], selecting option 6.RAK2247(USB).
6. After the installation is completed, after a while, i can't connect to it anymore and have to format the Raspberry.

Disclaimer: Sorry for the long post, i tried to give as much info as possible.

SWoto, 2020-10-06:
❗ One important thing that I found out is that (after configuring the gateway_id) even though I can't connect to the gateway, TTN is seeing it.

[image]

However, this is really problematic, because I can't change anything in the gateway this way.

Scobber, 2020-10-06:
does a packet forward though the gateway in the data tab in TTN

that last seen counter is the result of a STAT data format, in a earlier release the 'temp' variable was packaged in the stat packet. The Things Network did not support this, and silently discarded the packet. The GPS data is part of the stat packet so the result was no co-ordinates of the gateway on received packets, but it still incremented the last seen.

I have a utility I can post a link to, which is essentially a GWP proxy written in python. It may provide you some clues on what's going wrong.

Scobber, 2020-10-07:
If you are interested in the GWP proxy avenue to try and track down what's going wrong, see https://thescobber.com/2020/10/07/proxying-gwp-packets/
https://github.com/Scobber/GWP-Proxy/blob/master/README.md

If anyone wants to make any contributions to it I don't mind.

SWoto, 2020-10-08:
Hello @Scobber, thank you for replying. 

I'll run another test today and if it results in another failure I'll use your suggestion and post the results here.

SWoto, 2020-10-09:
Problem related to insufficient power. Solved by changing Power Source.

closed 2020-10-09 by SWoto