RAKWireless/rak_common_for_gateway #43: Install doesn't work for RAK7248/RAK2287

TakahashiNZ, 2021-02-10:
Hi, I'm trying to get a RAK2287 plugged into the pi hat working.
As there is no option for a plain RAK2287 I choose the RAK7248 no LTE option.
Install completed OK but the concentrator fails to start.
I suspect the install script has an bug as it appears to install the SX1301 packet forwarder from the RAK7243 repository rather than the SX1302 packet forwarder.

Any suggestions?

thanks

cstratton, 2021-02-10:
As far as I can see the rak2287_spi directory is the only one with sx1302 support, eg the only one compatible with the rak2287

The latest commit in that directory happens to also mention that it is a fix for the RAK7248 platform.

So try that.  You should at least be able to get the LoRa aspect working, the GPS might be more complicated depending if either the UART or I2C is actually pinned out on the board you're putting the mPCIe card into.

TakahashiNZ, 2021-03-10:
Sorry for the delay in my reply, I was sidetracked on some other projects.

I was able to get the RAK2287 working by cloning the rak_common repo and then running the install.sh in the RAK2287_spi folder. I also had to create a global_conf.json and local_conf.json.

Now I'm trying to get the GPS working. I2c is hooked up to my Pi, but I keep getting a permission error:

WARNING: [main] impossible to open /dev/i2c-1 for GPS sync (check permissions)

This seems a bit odd as sudo doesn't help and I have added pi user to i2c group and I can run i2cdetect without sudo.

cstratton, 2021-03-10:
I'd suggest you build with debug enabled (add -g to the Makefile) and debug your way through https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak2287_spi/loragw_gps.c or at least try firing up the packet forwarder under strace to see if you can figure out what specifically is wrong.

rgillan, 2021-03-12:
So we've taken a different approach because we kept getting stuck at the same point, and we're trying to build on an Armbian based arm board so not sure if this helps.  We have downloaded the sx1302_hal (https://github.com/Lora-net/sx1302_hal) and overwritten the three files it appears have changed: packet_forwarder/src/lora_pkt_fwd.c, libloragw/src/loragw_gps.c and libloragw/src/loragw_stts751.c.  If you then follow the install instructions in the Semtech repo you should be able to get the stand alone rak2287 running.  We also had to edit the hard coded (seriously?) i2c reference in the rak version of loragw_gps.c line 423)
at least we have our 2287 up and running now ;)

TaylorIsAGoodboy, 2021-09-06:
> Now I'm trying to get the GPS working. I2c is hooked up to my Pi, but I keep getting a permission error:

If you want to use I2c gps, you should buy pi-hat(i2c gps) for RAK2287. The default pi-hat version is pi-hat(uart gps) for RAK2287.

Choose the "RAK7248 no LTE (RAK2287 SPI + raspberry pi) "option, use uart gps.  
you can use 'stty -F /dev/ttyAMA0 -echo ispeed 9600 && cat </dev/ttyAMA0' command to check whether GPS is work or not.

If use I2c gps,  you may choose 'RAK7248 with LTE (RAK2287 SPI + LTE + raspberry pi)'

vincegre, 2021-10-04:
Are all these problems still present to get RAK2287 SPI version with GPS working ? As I'm struggling to get mine (RAK2287 SPI with GPS board mounted on RAK2287 Hat + Pi4) working on TTN ! I tried first the image supplied by RAK but doesn't succeed to connect at TTN (I have changed in json config file URL to point at eu1...) although the Lora board is working when I use it in Chirpstack mode !
I then tried the manual way with pi debian install then clone repo here and start the setup script but I end up with a system that is not working ! the start script call some stuffs about SX1301 instead of 1302 :(

closed 2021-10-09 by TaylorIsAGoodboy