RAKWireless/rak_common_for_gateway #46: RAK 2247 USB ERROR: [main] failed to start the concentrator

mo-norant, 2021-03-04:
When I restart the concentrator after the first time. This error is shown. I use the RAK2247 USB dongle gateway

ERROR: [main] failed to start the concentrator

[image]

ZhuQI77, 2021-03-19:
@mo-norant 
Can you see the following devices through the `lsusb `command?
[image]

If you can see it, maybe you need to pull down the voltage of pin 22, for example, connect pin 22 to ground.

Peter-Eismann, 2022-10-10:
Hi,

I just got my RAK2247 (the SPI version) and a USB-Adapter. But when I plug it in (no matter on a Windows or Linux PC) it stays dead. The power LED on the RAK is on but it's not recognized by the system. lsusb does not show it. I tested the USB dongle with another 3G card and that works fine. Is there something I have to do, to make the SPI version compatible to the USB dongle?