RAKWireless/rak_common_for_gateway #58: Extremly poor performance RAK5146(USB) / RAK5146-226

nmaas87, 2021-09-12:
I have been trying to use the RAK5146-226 both on a generic WWAN USB breakout as well as on a balenaFin with multiple CubeCells. However, the connection is extremely unstable - I cannot send a simple 01020304 relaibly, running it over night showed that FCT skips dozens and sometimes hundereds of packets. Also, the reset script for the RAK5146 ( https://github.com/RAKWireless/rak_common_for_gateway/blob/master/lora/rak5146/reset_lgw.sh ) contains pulling down GPIO pins - which are not available while using the RAK5146-226 (USB, GPS, LBT version) - so I guess this is part of the problem. When will this be fixed and an official firmware be ready, like for the RAK2287 or othr products?

TaylorIsAGoodboy, 2021-09-14:
We will test it and make a official firmware ASAP.

TaylorIsAGoodboy, 2021-09-16:
I made a RAK5146-USB firmware, download url is https://downloads.rakwireless.com/LoRa/RAK5146/Firmware/.
You need to change server_addr of global_conf.json by yourself.  And RAK5146-USB has no need to execute reset_lgw.sh which is for spi version.

By the way, you should confirm the RX2 SF when you create gateway and device on TTN v3.

nmaas87, 2021-09-16:
Thank you Taylor,
I normally use the "Europe 863-870 MHz (SF9 for RX2 - recommended)" for both the CubeCells as well as the RAK5146.
I hope thats correct? (I use EU 868 MHz hardware for both the gateway as well as the CubeCells)
Do I need to configure this in some way in the Firmware? (or anything else than the server_addr to be changed?)

TaylorIsAGoodboy, 2021-09-16:
[image]

TaylorIsAGoodboy, 2021-09-16:
If you choose other options, no need to change change gateway configuration, but you need to change your device node configuration.  For convenience, you'd beeter choose "RX2 SF12" option.

closed 2021-09-27 by TaylorIsAGoodboy