RAKWireless/rak_common_for_gateway #70: systemd-timesyncd.service not working

basking-in-the-sun2000, 2021-12-25:
Trying to install rak common, but fails with a **Failed to set ntp: NTP not supported**

Tried reinstalling systemd-timesyncd, but the install process for rak common. After reinstalling timesyncd, but before rak common

```
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-12-25 01:44:16 UTC; 28s ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 20659 (systemd-timesyn)
     Status: "Initial synchronization to time server 172.104.13.118:123 (0.debian.pool.ntp.org)."
      Tasks: 2 (limit: 4915)
        CPU: 113ms
     CGroup: /system.slice/systemd-timesyncd.service
             └─20659 /lib/systemd/systemd-timesyncd

Dec 25 01:44:16 rak-gateway systemd[1]: Starting Network Time Synchronization...
Dec 25 01:44:16 rak-gateway systemd[1]: Started Network Time Synchronization.
Dec 25 01:44:16 rak-gateway systemd-timesyncd[20659]: Initial synchronization to time server 172.104.13.118:123 (0.debian>

```

If I check after installing rak common, it returns

systemctl status systemd-timesyncd.service 
```
● systemd-timesyncd.service
     Loaded: masked (Reason: Unit systemd-timesyncd.service is masked.)
     Active: inactive (dead) since Sat 2021-12-25 01:46:23 UTC; 32s ago
   Main PID: 20659 (code=exited, status=0/SUCCESS)
     Status: "Idle."
        CPU: 117ms

Dec 25 01:44:16 rak-gateway systemd[1]: Starting Network Time Synchronization...
Dec 25 01:44:16 rak-gateway systemd[1]: Started Network Time Synchronization.
Dec 25 01:44:16 rak-gateway systemd-timesyncd[20659]: Initial synchronization to time server 172.104.13.118:123 (0.debian>
Dec 25 01:46:23 rak-gateway systemd[1]: Stopping Network Time Synchronization...
Dec 25 01:46:23 rak-gateway systemd[1]: systemd-timesyncd.service: Succeeded.
Dec 25 01:46:23 rak-gateway systemd[1]: Stopped Network Time Synchronization.
```

basking-in-the-sun2000, 2021-12-25:
added the following after apt install ntp in sysconf/install.sh. Seems to have worked

```
systemctl unmask systemd-timesyncd.service
apt install -y systemd-timesyncd
```

TaylorIsAGoodboy, 2021-12-28:
Thank you, I will fix it.

closed 2022-01-14 by TaylorIsAGoodboy