RAKWireless/rak_common_for_gateway #71: RAK 2287 USB (current repo on RPi4B) - MCU version mismatch (expected: 00.02.06, got:V01.00.00) and a following ACK type error + read REQ_MULTIPLE_SPI ack

mars96, 2022-01-12:
I've got the RAK 2287 USB module here which is plugged into a Raspberry Pi 4B (4 MB RAM) on the latest Raspberry Pi OS.
I've tested the standard ARMHF as well as the ARM64 release (fresh installs).

So far I've not been able to make it work with the latest (current) github code available, while following the "Installation procedure" steps to the letter.

After writing the SDCard I enabled SPI, etc. as described in Step 2 of the "Installation procedure" for the "rak_common_for_gateway" repository.
Then I cloned the repo and ran the install.sh ("sudo ./install.sh --chirpstack=not_install" -- without the chirpstack backend as I have this running for month).

I've selected the gateway model as "9" ("9.RAK2287 USB") for "RAK2287 USB". Everything seems to be ok (compile,...) except for the errors thrown by the "ttn-gateway" when it is actually started.
The RAK2287 USB I have here has no GPS (at least there is no u.FL connector on the board to connect a GPS antenna).

Here is the rest of the log right before the first WARNING and the following ERROR lines (the full log can be found below):
```
[...]
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Connect to MCU
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: WARNING: MCU version mismatch (expected:00.02.06, got:V01.00.00)
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Concentrator MCU version is V01.00.00
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: MCU status: sys_time:1887454 temperature:-0.0oC
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: ERROR: received wrong ACK type (0xFF)
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: ERROR: failed to read REQ_MULTIPLE_SPI ack
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: ERROR: [main] failed to start the concentrator
Jan 12 18:31:48 rak-gateway systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
Jan 12 18:31:48 rak-gateway systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
[...]
```

Please help. What could be the issue?
Is there another configuation parameter I should know and change which is currently not handled by the script where I have selected the correct device (RAK2287 USB)?

Here is everything I can provide ...starting with the basic USB details...

**"sudb lsusb" output (looks perfect to me: 0483:5740 STMicroelectronics Virtual COM Port):**
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 05af:0906 Jing-Mold Enterprise Co., Ltd 2.4G Dongle
Bus 001 Device 003: ID 0483:5740 STMicroelectronics Virtual COM Port
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**"sudo dmesg | grep -i tty" output:**
```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:76:70:69 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=f514dbcb-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether
[    0.001810] printk: console [tty1] enabled
[    1.586317] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    3.302189] systemd[1]: Created slice system-getty.slice.
[    5.918326] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

In the following logs and config files I've removed a part of the gateway id which seems to be read from the device itself.

**Here is the relevant /var/log/syslog section...**
```
Jan 12 18:31:47 rak-gateway systemd[1]: Started The Things Network Gateway.
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: *** Packet Forwarder ***
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: Version: 2.0.1
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: *** SX1302 HAL library version info ***
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: Version: 2.0.1;
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: ***
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Little endian host
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: found configuration file global_conf.json, parsing it
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: com_type USB, com_path /dev/ttyACM0, lorawan_public 1, clksrc 0, full_duplex 0
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: antenna_gain 0 dBi
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Configuring legacy timestamp
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: no configuration for SX1261
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
Jan 12 18:31:48 rak-gateway ttn-gateway[5532]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
[... 147 lines cut ...]
        "beacon_freq_nb": 1,
        "beacon_freq_step": 0,
        "beacon_datarate": 9,
        "beacon_bw_hz": 125000,
        "beacon_power": 27
    },

    "debug_conf": {
        "ref_payload":[
            {"id": "0xCAFE1234"},
            {"id": "0xCAFE2345"}
        ],
        "log_file": "loragw_hal.log"
    }
}
```

**Here is the /opt/ttn-gateway/packet_forwarder/lora_pkt_fwd/local_conf.json to set the actual GW-ID :**
```
{
        "gateway_conf": {
                "gateway_ID": "E45F01FFFE[........]"
        }
}
```

TaylorIsAGoodboy, 2022-01-14:
Hello, I think I know where the problem is. 
The reason is the firmware of USB to spi chip is mismatch.
Could you please select 'RAK5146 USB' in "Installation procedure" to check if it works ok.

mars96, 2022-01-14:
Hi Taylor,
thank you for the hint. I'll check it and report the result.

mars96, 2022-01-14:
Here the update on Taylors hint.

When I select "RAK5146 USB" (SX1303) instead of the actual "RAK2287 USB" (SX1302) the code compiles and the service will start.
It does NOT run into the "MCU version mismatch (expected:00.02.06, got:V01.00.00)" issue.

**Now the BUT...**
I kept the Gateway ID and my server (ChirpStack backend) did receive the gateway pings but it didn't show actual "Live LoRaWAN Frames" (as it did for example for the "RAK2247 USB" before which used the same Gateway ID).
Then I added a new Gateway with a modified Gateway ID to ChirpStack and changed the Gateway ID in the local config. After I restarted the service the ping did work and  "Live LoRaWAN Frames" did again show up.
This is a little bit strange....I'll look into it.
I also need to check how it handles actual LoRa devices....

**Anyway here is the relevant part of the "/var/log/syslog" output:**
```
Jan 14 10:29:03 rak-gateway systemd[1]: systemd-fsckd.service: Succeeded.
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: *** Packet Forwarder ***
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: Version: 2.1.0
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: *** SX1302 HAL library version info ***
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: Version: 2.1.0;
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: ***
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Little endian host
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: found configuration file global_conf.json, parsing it
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: com_type USB, com_path /dev/ttyACM0, lorawan_public 0, clksrc 0, full_duplex 0
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: antenna_gain 0 dBi
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Configuring legacy timestamp
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: SX1261 spi_path is not configured in global_conf.json
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: radio 0 enabled (type SX1250), center frequency 867500000, RSSI offset -215.399994, tx enabled 1, single input mode 0
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: radio 1 enabled (type SX1250), center frequency 868500000, RSSI offset -215.399994, tx enabled 0, single input mode 0
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 5 to 12
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7, Explicit header
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: gateway MAC address is configured to AA555A0000000000
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: server hostname or IP address is configured to "192.168.20.44"
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: upstream port is configured to "1700"
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: downstream port is configured to "1700"
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: downstream keep-alive interval is configured to 10 seconds
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: statistics display interval is configured to 30 seconds
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: upstream PUSH_DATA time-out is configured to 100 ms
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: packets received with a valid CRC will be forwarded
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: packets received with a CRC error will NOT be forwarded
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: packets received with no CRC will NOT be forwarded
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: GPS serial port path is configured to ""
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Reference latitude is configured to 0.000000 deg
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Reference longitude is configured to 0.000000 deg
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Reference altitude is configured to 0 meters
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Beaconing period is configured to 0 seconds
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: Beaconing signal will be emitted at 869525000 Hz
[... cut ...]
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: ARB: dual demodulation disabled for all SF
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: [main] concentrator started, packet can now be received
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: concentrator EUI: 0x0016c0...........
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: [down] PULL_ACK received in 0 ms
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: [down] PULL_ACK received in 0 ms
Jan 14 10:29:27 rak-gateway ttn-gateway[742]: INFO: [down] PULL_ACK received in 0 ms
[...]
```

closed 2022-07-18 by TaylorIsAGoodboy