RAKWireless/rak_common_for_gateway #77: RAK2245 Pi Hat + Raspberry Pi 4 - Access Point

Karol-Karas, 2022-01-20:
After enabling AP mode it does not appear in the list of available networks. It worked perfectly in previous version (RAK7243_on_Raspbian_OS_V4.2.5_20200909.img). I've already set localisation and tried to change SSID and password but it didn't help.

I can't see any errors while creating AP Service.  Would you mind to take a look? ✌

TaylorIsAGoodboy, 2022-02-09:
Hello, does the problem still exist?  Please show me the content of **/usr/local/rak/gateway-config-info.json**

Karol-Karas, 2022-02-09:
@TaylorIsAGoodboy 

Yep, there's still problem with AP mode ( :( ), but Client Mode works perfectly. 

```json
{
        "lora_server":
        {
"server_plan":"1",
                "lora_server_ip":"127.0.0.1",
                "up_port":"1700",
                "down_port":"1700",
                "loraserver_adr":"2",
                "freq":"5"
        },
        "wifi":
        {
"wifi_mode":"2",
                "wifi_ip":"",
                "wifi_gw":"",
"ap_ssid":"myssid",
"ap_pwd":"mypwd"
        },
        "lan":
        {
                "lan_ip":"192.168.10.10",
                "lan_gw":"192.168.10.1"
        },
        "apn":
        {
                "apn_name":"HOLOGRAM",
                "apn_baud":"115200"
        },
        "time":
        {
                "sync_time":0
        },
        "active_lora_server":"2",
        "install_img":"0",
"install_chirpstack":"1",
        "install_lte": "0",
        "lte_mode":"2"
}
```

TaylorIsAGoodboy, 2022-02-10:
Did you use `sudo gateway-config` command to configure wifi, Or you used `sudo raspi-config`?

Karol-Karas, 2022-02-10:
gateway-config, I used raspi-config to enable I2C and SPI interface etc.

TaylorIsAGoodboy, 2022-02-11:
You can add touch a file named rak_ap at /boot，then reboot, the configure  of wifi will be reset to factory.
please have a try.

Karol-Karas, 2022-02-19:
Nope, doesn't work. Nevermind, I will just use dhcpcd.conf. Thanks for help!

closed 2022-02-19 by Karol-Karas