RAKWireless/rak_common_for_gateway #81: RAK 2287 + PI Errors running forwarder

fuzemobi, 2022-05-01:
Hi, I'm struggling to get my default pi version working either from your image download or from this repo.  The json configs don't appear to be parsing correctly.   Is it possible that the locale and/or the device selection are incorrect?

Hardware: Raspberry Pi 4 + Rak2287 + Pi Hat
Image: Latest Raspberry Pi OS Lite (32-bit) 
Released 2022-04-04 

On install, the model selection was:
*	 7.RAK7248(SPI) no LTE (RAK2287(SPI) + raspberry pi)

After following all steps in the readme, I can't seem to get the forwarder to run.  A couple of errors that stand out:
- May  1 17:23:31 rak-gateway-01 ttn-gateway[1647]: /opt/ttn-gateway/packet_forwarder/lora_pkt_fwd/start.sh: line 6: echo: write error: Device or resource busy
- May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
- May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: ERROR: Failed to set SX1250_0 in STANDBY_RC mode
- May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters

LOG dump:
root@rak-gateway-01:~# tail -f /var/log/syslog | grep ttn-gateway
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: INFO: packets received with no CRC will NOT be forwarded
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: This is uart for GPS.
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: INFO: [main] TTY port /dev/ttyAMA0 open for GPS synchronization
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: Opening SPI communication interface
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: Note: chip version is 0x00 (v0.0)
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: ERROR: Failed to set SX1250_0 in STANDBY_RC mode
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: ERROR: failed to setup radio 0
May  1 17:23:26 rak-gateway-01 ttn-gateway[1631]: ERROR: [main] failed to start the concentrator
May  1 17:23:26 rak-gateway-01 systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
May  1 17:23:26 rak-gateway-01 systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
May  1 17:23:31 rak-gateway-01 systemd[1]: ttn-gateway.service: Scheduled restart job, restart counter is at 37.
May  1 17:23:31 rak-gateway-01 ttn-gateway[1647]: /opt/ttn-gateway/packet_forwarder/lora_pkt_fwd/start.sh: line 6: echo: write error: Device or resource busy
May  1 17:23:32 rak-gateway-01 ttn-gateway[1657]: CoreCell reset through GPIO17...
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: *** Packet Forwarder ***
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: Version: 2.0.1
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: *** SX1302 HAL library version info ***
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: Version: 2.0.1;
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: ***
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Little endian host
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: found configuration file global_conf.json, parsing it
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: global_conf.json does contain a JSON object named SX130x_conf, parsing SX1302 parameters
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: com_type SPI, com_path /dev/spidev0.0, lorawan_public 1, clksrc 0, full_duplex 0
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: antenna_gain 0 dBi
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Configuring legacy timestamp
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: no configuration for SX1261
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Configuring Tx Gain LUT for rf_chain 0 with 16 indexes for sx1250
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: radio 0 enabled (type SX1250), center frequency 904300000, RSSI offset -215.399994, tx enabled 1, single input mode 0
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: radio 1 enabled (type SX1250), center frequency 905000000, RSSI offset -215.399994, tx enabled 0, single input mode 0
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 5 to 12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8, Explicit header
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: FSK channel 8 disabled
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: gateway MAC address is configured to AA555A0000000000
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: server hostname or IP address is configured to "127.0.0.1"
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: upstream port is configured to "1680"
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: downstream port is configured to "1680"
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: downstream keep-alive interval is configured to 10 seconds
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: statistics display interval is configured to 30 seconds
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: upstream PUSH_DATA time-out is configured to 100 ms
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with a valid CRC will be forwarded
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with a CRC error will NOT be forwarded
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with no CRC will NOT be forwarded
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: GPS serial port path is configured to "/dev/ttyAMA0"
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Reference latitude is configured to 0.000000 deg
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Reference longitude is configured to 0.000000 deg
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Reference altitude is configured to 0 meters
[... 4 lines cut ...]
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Beaconing datarate is set to SF12
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Beaconing modulation bandwidth is set to 500000Hz
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: Beaconing TX power is set to 27dBm
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: global_conf.json does contain a JSON object named debug_conf, parsing debug parameters
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: got 2 debug reference payload
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: reference payload ID 0 is 0xCAFE1234
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: reference payload ID 1 is 0xCAFE2345
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: setting debug log file name to loragw_hal.log
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: found configuration file local_conf.json, parsing it
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: gateway MAC address is configured to B827EBFFFEF76797
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with a valid CRC will be forwarded
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with a CRC error will NOT be forwarded
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: INFO: packets received with no CRC will NOT be forwarded
[... cut ...]
May  1 17:23:34 rak-gateway-01 ttn-gateway[1655]: ERROR: [main] failed to start the concentrator
May  1 17:23:34 rak-gateway-01 systemd[1]: ttn-gateway.service: Main process exited, code=exited, status=1/FAILURE
May  1 17:23:34 rak-gateway-01 systemd[1]: ttn-gateway.service: Failed with result 'exit-code'.
May  1 17:23:39 rak-gateway-01 systemd[1]: ttn-gateway.service: Scheduled restart job, restart counter is at 38.
May  1 17:23:39 rak-gateway-01 ttn-gateway[1671]: /opt/ttn-gateway/packet_forwarder/lora_pkt_fwd/start.sh: line 6: echo: write error: Device or resource busy

Scobber, 2022-05-02:
did you setup the pi correctly?

from memory there is some stuff to do in sudo raspi-config
the likes of enabling SPI and such

fuzemobi, 2022-05-03:
I have included the configuration files for the rak and a picture of the setup.  All of the instructions from the RAK setup/github page have been set including the raspi-config settings.  

The main question I have is the selection of the rak model when running ./install.sh from the ~/rak_common_for_gateway/install.sh
For this build I have tried several settings 7.RAK7248(SPI) no LTE (RAK2287(SPI) + raspberry pi) as well as 9.RAK2287(USB). 

[rak_gw_model.json.txt](https://github.com/RAKWireless/rak_common_for_gateway/files/8611257/rak_gw_model.json.txt)
[local_conf.json.txt](https://github.com/RAKWireless/rak_common_for_gateway/files/8611258/local_conf.json.txt)
[global_conf.json.txt](https://github.com/RAKWireless/rak_common_for_gateway/files/8611259/global_conf.json.txt)
[gateway-config-info.json.txt](https://github.com/RAKWireless/rak_common_for_gateway/files/8611260/gateway-config-info.json.txt)
[syslog.txt](https://github.com/RAKWireless/rak_common_for_gateway/files/8611261/syslog.txt)
[image]

TaylorIsAGoodboy, 2022-07-18:
Hi @fuzemobi , your should select `9.RAK2287(USB)`.
If you use pi-hat connect RAK2287(USB), after connect pi-hat to pi, you also need a micro usb cable to connect pi-hat to usb of pi.