RAKWireless/rak_common_for_gateway #89: Failed to disable unit: Unit file hciuart.service does not exist.

guanbo123456, 2023-04-18:
I am installing rak_common_for_gateway for RAK5146 on Ubuntu 20.04.4 LTS (GNU/Linux 4.19.206 aarch64), but there is a error.
Failed to disable unit: Unit file hciuart.service does not exist.
The following is my steps.
How can I solve it?

forlinx@rak-gateway:~/rak_common_for_gateway$ sudo ./install.sh
Please select your gateway model:
*        1.RAK2245
*        2.RAK7243/RAK7244 no LTE
*        3.RAK7243/RAK7244 with LTE
*        4.RAK2247(USB)
*        5.RAK2247(SPI)
*        6.RAK2246
*        7.RAK7248(SPI) no LTE (RAK2287(SPI) + raspberry pi)
*        8.RAK7248(SPI) with LTE (RAK2287(SPI) + LTE + raspberry pi)
*        9.RAK2287(USB)
*        10.RAK5146(USB)
*        11.RAK5146(SPI)
*        12.RAK5146(SPI) with LTE
Please enter 1-12 to select the model:11
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Hit:6 https://artifacts.chirpstack.io/packages/4.x/deb stable InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
38 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'restricted/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'universe/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'multiverse/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'restricted/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'universe/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'multiverse/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'restricted/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'universe/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'multiverse/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-security InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'restricted/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-security InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'universe/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-security InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'multiverse/binary-armel/Packages' as repository 'http://ports.ubuntu.com/ubuntu-ports focal-security InRelease' doesn't support architecture 'armel'
N: Skipping acquire of configured file 'main/binary-armel/Packages' as repository 'https://artifacts.chirpstack.io/packages/4.x/deb stable InRelease' doesn't support architecture 'armel'
/home/forlinx/rak_common_for_gateway/rak /home/forlinx/rak_common_for_gateway
Failed to disable unit: Unit file hciuart.service does not exist.
forlinx@rak-gateway:~/rak_common_for_gateway$ uname -a
Linux rak-gateway 4.19.206 #3 SMP Thu Dec 1 21:51:04 PST 2022 aarch64 aarch64 aarch64 GNU/Linux
forlinx@rak-gateway:~/rak_common_for_gateway$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

cboy1422081986, 2023-06-05:
我也有同样的问题

GitUser8675309, 2024-04-06:
I have the same issue. Has anybody got this to work? It seems I've tried everything.

orangepi@LoRaWAN:~$ cd ~/rak_common_for_gateway
sudo ./install.sh
[sudo] password for orangepi:
Please select your gateway model:
*        1.RAK2245
*        2.RAK7243/RAK7244 no LTE
*        3.RAK7243/RAK7244 with LTE
*        4.RAK2247(USB)
*        5.RAK2247(SPI)
*        6.RAK2246
*        7.RAK7248(SPI) no LTE (RAK2287(SPI) + raspberry pi)
*        8.RAK7248(SPI) with LTE (RAK2287(SPI) + LTE + raspberry pi)
*        9.RAK2287(USB)
*        10.RAK5146(USB)
*        11.RAK5146(SPI)
*        12.RAK5146(SPI) with LTE
Please enter 1-12 to select the model:11
Hit:1 http://repo.huaweicloud.com/debian bookworm InRelease
Hit:2 http://repo.huaweicloud.com/debian bookworm-updates InRelease
Hit:3 http://repo.huaweicloud.com/debian bookworm-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
/home/orangepi/rak_common_for_gateway/rak /home/orangepi/rak_common_for_gateway
Failed to disable unit: Unit file hciuart.service does not exist.
orangepi@LoRaWAN:~/rak_common_for_gateway$ ^C

rth, 2025-08-16:
As far as I found, hciuart is specific for Raspberry PI / Raspbian. So if you use a different OS, it should be OK to skip that. 

You can comment out the line 
```
systemctl disable hciuart
```
in install.sh

MoandDi, 2025-11-14:
I know that this is so incredibly late, but I've had the same issue and was able to find a solution through some forum diving. Hopefully this helps someone else with the same problem. The underlying issue is that the Bluetooth (hciuart) service is not installed by default, and therefore cannot be disabled. This is enough to halt the entire install. All you have to do is install the Bluetooth service (so that the installer can disable it)

`sudo apt-get install pi-bluetooth`

Just run this and then try the installer again, should work.